RoBorregos/robocup-home #31: Define a new organization for ROS Packages

diegocardozo97, 2020-04-19:
To achieve a better organization and to even allow externals easily use our ROS packages without having to change names, define and implement a new packages' name convention.

I think is fine having that convention and you (navigation, with this PR) can start doing it like that. About the mentioned problem, I think could be later resolved with a new pkg name and path standard like: `catkin_home/src/robo/athome/navigation/a_pkg` -> `package(robo_athome_navigation_apkg)`.

_Originally posted by @diegocardozo97 in https://github.com/RoBorregos/Robocup-Home/pull/25#issuecomment-615487710_

paulvazbad, 2020-04-29:
### Changes
I've been thinking about this subject, and I propose the following changes:

1.  Add RB_ as a prefix to all the packages defined/developed by us. For example:
_main_engine_ -> _RB_main_engine_

2. Move the parser from action_selectors to main_engine, as I believe (and I think @diegocardozo97 agrees with me), the parser is more involved with the action manager than with the stt. 

3. Rename the following packages:
    -action_selectors -> RB_speech_to_text
  
 4. Make a new package called RB_vision and move ObjectDetector from the action_selectors to this new packages.

5. Include @IquiBalamHM 's code in RB_vision (#35)

What do you guys think?

AuroTB, 2020-05-02:
Hi! I agree with Paul to just add the "rb_" prefix (or maybe change it to a more specific one like roborregos_, robert_, etc), just taking into account the [package convention](https://www.ros.org/reps/rep-0144.html) of having it lowercase and with a significant, distinguishing characteristic.

> - if a package is specialized for a software project, prepend its name
> - if a package is specialized for a hardware piece, prepend its name
> - if a package is **specialized for a robot, prepend its name**
> - if a package is specialized by an entity (lab, company, ...), prepend the name of the entity. Once the package is commonly used, owned and maintained, that name can be dropped

You should check it out, because there is also a section for [special cases](https://www.ros.org/reps/rep-0144.html#special-cases) such as launch files packages and bring-up files, which is exactly what I navigation is trying to implement and we'll have to implement in the future.

paulvazbad, 2020-05-04:
Thats amazing, thanks for the info!
So I guess the rb_ (robert_) is decided then?
What do you think about the other changes?

diegocardozo97, 2020-05-04:
That info, Aurora, is very nice, I didn't know it. I think something like rb_robert_ or rb_home_ is good.
Additionally, I was wondering that if we in the future going to have too many packages, we should start using like java's convention for the packages' name.

For example,
```
catkin_home
   |--->src
        |--->rb
        |       |--->home
        |       |     |--->move
        |       |     |     |--->pid
        |       |     |     |       |--->~actual package (rb_home_move_pid)~
        |       |     |     |--->state
        |       |     |            |--->~actual package (rb_home_move_state)~
        |       |     |--->mainengine
        |       |          |--->voice
        |       |          |       |--->~actual package (rb_home_mainengine_voice)~
        |       |          |--->state
        |       |                 |--->~actual package (rb_home_mainengine_state)~
        |       |--->vsss
        |            |--->vision
        |                  |--->colortracking
        |                         |--->~actual package (rb_vsss_vision_colortracking)~
        |--->third_party
               |--->vizbox
                      |--->~actual package (vizbox)~
```
Then, by including the path in the package name, we can avoid collisions of the packages, for example move's state and mainengine's state. 
Maybe is too much, but is an option. I say that ROS was made originally for research purpose, that's why is too _bad_ for actual code.

AuroTB, 2020-05-05:
All right, I think I like Cardozo's idea to go with the prefix <rb_home_area> for the @HOME project, where _area_ refers to the folder and the area where this package is being used.
So, to summarize, from now on we'll have to change the packages names according to this:

- **All packages:** use prefix <rb_home_ _area_>
- **General area launch file package:** use suffix <_launch>
- **Global launch file package:** use suffix <_bringup>
- **Meta package:** use suffix <_robot> or <_robert>
-**Packages containing URDFs and meshes:** use suffix <_description>
-**Package contains any of a ROS message/service/action:** use suffix <_msgs>

Did I miss something? We should add a small page to the wiki regarding this issue and update our own folder areas.

AuroTB, 2020-05-17:
Rules & style proposal: https://docs.google.com/document/d/1N8xFicfTHTpQcqkreHXX42Bds1BwUN9iJZjaGxtYXlQ/edit#

tomvik, 2020-10-08:
Will this be a thing?

tomvik, 2020-10-20:
@AuroTB @paulvazbad friendly ping

tomvik, 2020-10-27:
@AuroTB @paulvazbad 
[image]