RoBorregos/robocup-home #40: Obtener la Imagen en 3D A partir de la imagen en 2D.

paulvazbad, 2020-09-01:
Obtener en la imagen de 3D (RGBD) los puntos correspondientes a los píxeles de la imagen en 2D de las esquinas que delimitan al objeto.

tomvik, 2020-09-12:
@paulvazbad this is being blocked mainly by PR #35 
Also, by the fact that @diegocardozo97 hasn't uploaded his branch with his experiments with the camera, which could help as ground work for us.

paulvazbad, 2020-09-13:
I have reviewed PR #35 have you talked to @diegocardozo97? I can ping him too if needed.

tomvik, 2020-09-15:
Now that the PR #35 has been merged, these are the next steps:
1. Check Cardozo's [vision_intel_segment](Robocup-Home/tree/vision_intel_segment) branch and merge whatever is useful
2. Refactor the existing arm_vision package to follow the C++ style guide

paulvazbad, 2020-09-22:
Are you going to work on both things for the remaining time of this sprint?

tomvik, 2020-09-22:
That's the plan, but I'm unsure on how much I can complete.

tomvik, 2020-09-29:
As a follow up:
1. I'm checking Cardozo's branch, soon I should be able to give a LGTM to submit that PR.

2. The refactoring of the existing package is something that @IquiBalamHM was checking, according to our last meeting.

tomvik, 2020-10-08:
Currently working on refactoring Cardozo's code with the new Python standard.

closed 2020-10-13 by paulvazbad