RoBorregos/robocup-home #43: Draft de las medidas + diseño generico

paulvazbad, 2020-09-01:
Investigar/Hacer algún tipo de estudio para determinar la posición más activa de la pantalla en el robot. Esto involucra integrar con el diseño actual de Aldo y Amin.

Draft de las medidas de los componentes involucrados

Deliverable: Lista, sketch, de las medidas y componentes que afect ala cabeza + pantalla.

paulvazbad, 2020-09-15:
- Made CAD of the central microphone.
- Determined the optimal position for the screen.
-  Focussing on the face and head.

Deliverable links:

closed 2020-09-15 by paulvazbad

paulvazbad, 2020-09-17:
@GabrielaMl please add the links to the evidence/deliverables as a comment to this issue.

GabrielaMl, 2020-09-17:
https://tecmx-my.sharepoint.com/:f:/g/personal/a00825779_itesm_mx/Evrr53hF_7ZPtyLyytkLh-YBqMsqpydhdm7eCWHFch57jg?e=OIB87i