RoBorregos/robocup-home #55: Set up of Kinect in ROS

paulvazbad, 2020-09-15:
Migrating from Intel camera to Kinect.

Deliverable: PR

tomvik, 2020-09-15:
@IquiBalamHM and I were discussing briefly about the following two ideas:
1. Maybe it is possible to record a video or a set of photos with Intel's camera so he can use that for processing. More research needs to be done in order to find out if that's possible.
2. Either way, whether we're using the Kinect, Intel's camera, or any other camera, it would be nice to have the code in such a way that the **meat** of the code is immutable to the camera. Perhaps the processing, once the point cloud is extracted, could be the same for all cameras, or differing in some small parts.

paulvazbad, 2020-09-17:
I think related to the second point is the idea behind having the image capturing in the package devices/imageCapture.py . As of today, it is considered to have the processing completely separated from the image capture by subscribing to the topic `frames` that this node creates.

There is also a b
Take a look into [imageCapture.py](https://github.com/RoBorregos/Robocup-Home/blob/develop/catkin_home/src/devices/scripts/imageCapture.py)

There is still a TODO to implement this logic but the system architecture is ready to handle this change
`#Default webcam 
# 0 --- Integrated
# 2 --- Intel B/W
# 3 --- Intel Color
CAMERA = 0`

tomvik, 2020-09-29:
As mentioned on the issue #40 I believe @IquiBalamHM is currently working on refactoring the current arm_vision package to make it modular.

tomvik, 2020-10-20:
@IquiBalamHM can you write some update in here?

closed 2021-01-31 by Josecisneros001