RoBorregos/robocup-home #64: Gtest y continuos integration

paulvazbad, 2020-09-29:
Find a way to automate building and testing the develop branch
Deliverable: testing examples in the main engine package and docs on how to CI the project

tomvik, 2020-10-06:
@paulvazbad Maybe for the CI, could be as simple as using Github Workflows. A working example is in [RoBorrego's webpage](https://github.com/RoBorregos/roborregos-web/blob/develop/.github/workflows/node.js.yml)

The result is that everytime anyone makes a PR to develop, basically these two commands will run:
```
yarn install
yarn test
```
* It looks like this when you make a PR (ignore the blocking due to a lack of reviewers)
[image]

* One can go and check more info about the pipeline
[image]

* Once it's finished, it looks like this
[image]

I think this could be adapted to be something like:
```
// install dependencies
catkin_make // to make sure it builds
// command(s) to perform style checking
// command to perform tests
```

paulvazbad, 2020-10-07:
and `catkin_make run_tests` can be used to perform the tests, I like your idea @tomvik !

paulvazbad, 2020-10-07:
TODO: update action to include unit testing and integration testing

tomvik, 2020-10-08:
Will open a second issue for the testing

closed 2020-10-08 by tomvik