RoBorregos/robocup-home #65: Torque calculation and security of the robot

paulvazbad, 2020-09-29:
- Check that the robot has the correct dimensions and that the torque in the elevator is wont make it fall over someone.
- Calculate how much does it take for the robot to reach the top.

paulvazbad, 2020-10-10:
@AminMendez @Lufergvera @AldoSamaniego please add to this issue updates on the subject:
Calculations done, modifications to the design, security risks detected, etc....

AminMendez, 2020-10-13:
https://docs.google.com/spreadsheets/d/1PbO1ljPmfABfIhJFhoARIgWsTb2fUlrairehncUXbyM/edit?usp=sharing

Analisis of Ball Screw and Rollover.

We found that with the current design the robot can experience Rollover when stopped inmediately at velocity of 0.7 m/s or higher. Due to the max velocity of the motor is 0.45 m/s, the mechanical team is going to change the design by adding an extra load at the back to lower the center of gravity or change the upper structure including the robot arm.

closed 2020-10-13 by paulvazbad