RoBorregos/robocup-home #68: Definir el estandar a seguir en python (PEP)

paulvazbad, 2020-10-02:
Como sugirio @tomvik hay que definir para los scripts de python que desarrollamos en @Home. Links utiles:
https://dev.to/j0nimost/setting-up-pep8-and-pylint-on-vs-code-34h
http://python.org/dev/peps/pep-0008/#variable-annotations
https://google.github.io/styleguide/pyguide.html#319-type-annotations

tomvik, 2020-10-02:
Do you think it's necessary to create a venv for this? It seems like the setup is straight forward: 
```
pip2 install pep8
pip2 install pylint
pip2 install autopep8
```

And then on the `.vscode/settings.json` to add the following lines:
```
"python.pythonPath": "/your/own/python/directory",
"python.formatting.provider": "autopep8",
"python.linting.pylintEnabled": true
```

So the above is only for normal linting and formatting. Now, the good thing is [pyright](https://github.com/microsoft/pyright):

Simply install the vscode extension, and add the following lines to the `.vscode/settings.json`:
```
"python.analysis.useLibraryCodeForTypes": true,
"python.analysis.typeCheckingMode":"strict"
```

And for instance, look at all the errors:
[image]

paulvazbad, 2020-10-06:
Looks great to me:) , Can you add a section to the wiki with this info?  I'll set it up in my pc and let you know how it goes

tomvik, 2020-10-06:
Done. 

The changes were:
* Added a wiki page: [Coding](https://github.com/RoBorregos/Robocup-Home/wiki/Coding)
* Modified the main wiki page to include the link to the Coding page
* Modified the sidebar to include the link to the Coding page

paulvazbad, 2020-10-07:
Awesome, thanks! Closing this now

closed 2020-10-07 by paulvazbad