RoBorregos/robocup-home #86: Test Migration to ROS Noetic

tomvik, 2020-10-27:
This may be a part 1 of 2, where the end goal is to move the project to ROS2.

Why go through ROS Noetic first? As recommended by ROS' webpage:

**This may be a good (optional) first step if you want to make the transition to ROS2 Foxy from ROS1 Kinetic or ROS1 Melodic. You can first port your ROS1 code to ROS1 Noetic and make sure everything works with Python3. Then, you can switch to ROS2.**

tomvik, 2020-10-27:
As talked with Paul privately; #85 is already functioning with:
- CI build and test
- main_engine

However, the rest of the scripts need to be tested for python3 compatibility and for that I'd need help of the people who wrote them or know how to run them @paulvazbad @AuroTB @RicardoChapaRomero

AuroTB, 2020-10-28:
Today we checked nav's action provider script and Nav Stack launch and everything ran successfully.
Other than that, Tommy had to install directly _navigation_, so that might have to be added to CI run actions (we'll have to make some tests).

paulvazbad, 2020-11-08:
I'm currently updating my Ubuntu from 18 to 20 and my ROS distribution from melodic to noetic. I'll let you know when I'm done so I can look into the python2 scripts that are outdated.
Side note: @Josecisneros001 mentioned that there are packages in speech that require python2, but he might have a 'hack' to fix that.

Josecisneros001, 2021-09-18:
Reference : #115

closed 2021-09-18 by Josecisneros001