RoBorregos/robocup-home #91: Change the motors

paulvazbad, 2020-11-03:
The robot's current motors are too slow with a velocity of approx 0.2 m/s, when the goal is of at least 0.43 m/s.
Adding @mechanics to the issue to help with the specs of the motoros.

Josecisneros001, 2020-11-03:
To get a speed of at least 0.43 m / s, we need some motors that are over 82 RPM.
(Pololu motor list)[https://www.pololu.com/category/116/37d-metal-gearmotors]
(I think this could be the one)[https://www.pololu.com/product/4755]
It would be great if any of you could calculate the torque required to move the robot.
@AminMendez @AldoSamaniego @BVDozen

Josecisneros001, 2020-11-03:
[This is the current motor](https://www.amazon.com/CQRobot-Encoder-Mounting-Bracket-80Kg-cm/dp/B0731MD6JL)

AminMendez, 2021-02-17:
The required torque to move the robot is a torque greater than the force of friction that go against the movement. Due to the design of the chasis this friction occurs between the motor shaft and it's cage. This friction coefficient is difficult to calculate theoritacally. We can estimate the value by knowing the actual weight and the force acting on the motor by applying the formula of equilibrium. To help our calculus we can use the torque vs speed chart of our current motor at nominal voltage 12V. If we apply 12V at motor and friction are not consider we expect that motor rotate at 40 rpm as the datasheet mention, but when load is applyied this rpm will reduce. The force acting against can be know by analyzing the chart and see the rpms the wheel does at a constant load. Once knowing the the force of friction we divide this value by the normal force, in this case the current weight of the robot.

[image]

Future design may include bearings to reduce this value of friction and improve robot movement. 

The motor propose by Cisneros is a good candidate for the replacement of the robot. with an initial acceleration of 0.952 m/s^2, max torque of 34 kgcm and max speed with no load of 100 rpms considering 70 kg of weight

Another candidate could be this one 
https://www.amazon.com/-/es/CQR37D64CPR131-3GR/dp/B0876CYVCC/ref=sr_1_1?__mk_es_US=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=cqrobot+131.3%3A1&qid=1613528682&sr=8-1

with an initial acceleration of 1.26 m/s^2, max torque of 45 kgcm and max speed with no load of 83 rpms

Josecisneros001, 2021-09-18:
Solve - Missing one motor.

closed 2023-11-03 by IvanRomero03