RoBorregos/robocup-home #96: Manufacture arm pulleys

paulvazbad, 2020-11-10:
Deliverable:
Photos and documentation.

Lufergvera, 2020-11-24:
[image]
[image]
[image]
[image]

AldoSamaniego, 2021-06-17:
[image]
[image]
[image]
[image]
[image]
[image]

closed 2023-11-03 by IvanRomero03