RoBorregos/robocup-home #100: Broken Build due to Github deprecation

tomvik, 2020-11-24:
Github has deprecated a set of commands used on the workflow that we use for our CI. 

Remembering that our CI depends on: https://github.com/betwo/github-setup-catkin to setup ROS.

More details on the deprecation of these commands is seen in: https://github.blog/changelog/2020-10-01-github-actions-deprecating-set-env-and-add-path-commands/

The solution I propose is to create an issue on betwo's repository, fork it, and solve it.

The error looks like:
[image]

paulvazbad, 2020-11-24:
I forked and made a hotfix:
https://github.com/paulvazbad/github-setup-catkin

tomvik, 2020-11-24:
Niceee! Do you know if we can use your forked branch in the meanwhile for our CI?

paulvazbad, 2020-11-25:
I believe we can use it

closed 2021-03-10 by Josecisneros001