RoBorregos/robocup-home #122: Top part of the elevator

AldoSamaniego, 2021-02-10:
Objectives:
Fix and make improvements of the top part of the elevator
Deliverables: 
SolidWorks CAD

Josecisneros001, 2021-03-10:
@AldoSamaniego @AminMendez take a look at this.

AldoSamaniego, 2021-06-17:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

BVDozen, 2021-10-02:
[image]
[image]
Soporte universal para micrófono utilizando velcro
Agujeros para display