RoBorregos/robocup-home #135: Iniciar Desarrollo Suspensión

Josecisneros001, 2021-02-24:
- Diseño de Suspensión para mejorar agarre

Entregable:
- [ ]  Investigación/Cálculos
- [ ] Modelaje Solidworks

AminMendez, 2021-03-10:
To begin the model of the robot suspension system first we need to calculate the teoretical values of the spring and damping constants that satisfy a smood displacement. Mathematical modeling of the four wheel system is complicated. We can simplify the sistem by only analysing one wheel as show below. 

[image]

A mass-spring damper system is model by the following equation: 

[image]

where m is the mass that holds a wheel, K the spring constant, b the damping coeficient and r(t) the external forces.

Using Laplace transform we can conclude:

[image]

[image]

The transfer function is given by:

[image]

Once having the model of the system we substitute the values we have, such as the mass. As we want a smood movement the 
spring must be a systm with critical damping, So the poles of the system must be equals:

s^2 +(b/m)*s + (k/m) = (s + p)^2 

giving us the value of b as function of k and m

b= 2* m* sqrt (k/m)

By using Matlab and WolframAlpha programs I model the system to give the reaction of any input to the system with knowing the constants.
[image]
[image]

My task of this week is to find the most suitable constant K for our robot, mantaining the desire stability.

AminMendez, 2021-04-21:
Continuing with the calculus of the suspension, We develop a MATLAB program in which by a given estability time we can know the values of the spring and damping constants. The first program is simulating a Step input with a variable magnitud and the second we can give the traslational function of the wheel.

UNIT STEP
[image]

Response with unit step magnitud of 10
[image]

Given Function
[image]
[image]

Response with Sin (t) input
[image]

Now by just given the Stability Time we can know the important constants  with a critical damping

AldoSamaniego, 2021-06-17:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

closed 2023-11-03 by IvanRomero03