RoBorregos/robocup-home #139: Gripper v1

Josecisneros001, 2021-03-10:
- Desarrollar v1.0 Gripper

Entregable
- [] Foto/Link Cad

AlonRuiz, 2021-03-24:
[image]
[image]
[image]
[image]

The parts and assembly shown above can be found in the shared OneDrive folder at: atHome>Brazo>Pinza.

Josecisneros001, 2021-04-07:
Working on:
- Servos Improvement

AlonRuiz, 2021-04-13:
Design report: Gripper Iterations

https://docs.google.com/document/d/12_VXpYMRWF4WLSRXYkXUYQPO0zy17tIpcvqSBW9r7Bc/edit?usp=sharing

AldoSamaniego, 2021-06-17:
[image]
[image]
[image]
[image]
[image]
[image]

closed 2021-09-18 by Josecisneros001